braindefender/KLP-Lame-Keycaps #1: 1.5U?

b166ar, 2023-08-12:
Any plans to make 1.5U thumb? For Chocofi and similar keyboards.

braindefender, 2023-08-13:
No, I do not plan to add 1.5U or other sizes. Perhaps, someone will make PR with this.

KLingO-MS, 2024-04-02:
Could You please create caps for index center column that will be tilted to the side (of course corner ones then would be tilted in 2 directions)

smallwat3r, 2024-05-18:
@b166ar I've found a MX version of a 1.5U thumb key on [this](https://github.com/vvhg1/clp-keycaps/blob/main/stl/CLP_Tilt7_1_5u%20v13.stl) repo. The keys have been inspired by the KLP Lame, so it fits quite well the design.

Because I'm using choc keycaps and not MX, I modified it so it has a choc plug, you can find the STL attached.

[CLP_Tilt7_1_5u_choc.stl.zip](https://github.com/braindefender/KLP-Lame-Keycaps/files/15366275/CLP_Tilt7_1_5u_choc.stl.zip)

[image]

closed 2024-11-06 by braindefender

braindefender, 2025-11-16:
Now there are 1.5 variants