braindefender/KLP-Lame-Keycaps #5: "combined" stl files are not being accepted by Jlcpcb as witten in README

DeBasti, 2023-09-03:
I tried to order the choc stl saddle combined https://github.com/braindefender/KLP-Lame-Keycaps/blob/master/STL/Choc/Combined/KLP-Lame-36-keys-Saddle-Combined.stl from jlcpcb and got the response that every stl file should only contain one part. I think the issue is that the keycaps are not connected thus being seen as different parts by jlcpcb.

Full response from jlcpcb:
Dear Sir/Madam, multiple parts (shells) were detected in your file, sorry to tell you that each file could only support 1 part (shell) for printing, please kindly separate the multiple shells into single files to place the order separately. Thanks !

braindefender, 2023-09-04:
Yes, I already addressed this issue and updated files. Also, there are limits to the width/height/file_size of the models, so I split combined parts into 18 caps per file. See this [commit](https://github.com/braindefender/KLP-Lame-Keycaps/commit/e47f24cc744540d3f8e0e0966f2e87a1adeed80d)

DeBasti, 2023-09-04:
Great! Thank you :)

braindefender, 2023-09-04:
Seems like JLC want only 10 caps per file. Working on it...

braindefender, 2023-09-04:
Now it is correct, I think

closed 2023-09-04 by braindefender