braindefender/KLP-Lame-Keycaps #6: Increase home keys nibs

mike1808, 2023-09-19:
The current version has a very small bumps to indicate home keys. When printed with 50 micron mSLA printer those nibs are very hard to feel. 
I tried increasing the ball diameter to 2mm and only having two of those and it feels good to use. 
[image]

I can provide pictures of printed caps if needed.

ald0o, 2023-09-27:
I printed them in FDM (TPU 95A) "super quality" (0.12 mm layers... so that would be 120 μm, comparing apples to apples!), and the bumps are also completely unnoticeable, as expected.

I am aware these keycaps were not meant for FDM, but except the issue with homing bumps, they are actually quite fine, so it would definitely be worth it to fix this little issue.

mike1808, 2023-09-28:
I tried making a photo but not sure you can see it. On the left is my modified version, on the right is the original version 
[image]

braindefender, 2023-10-27:
I'll fix it in the next update

braindefender, 2023-10-29:
You are welcome. Check out last commits and updates.

closed 2023-10-29 by braindefender

smallwat3r, 2024-04-29:
@mike1808, I quite like your version of the home keys with the increased two dots. Any chance you could share an STL file? Thanks!