braindefender/KLP-Lame-Keycaps #7: How to make the key caps transparent

sedghi, 2023-09-23:
I'm new to the world of custom ergo mech keyboards. I also know nothing about 3D printing.

My question is if i want to order on the manufacturer can i choose the transparent color?  Is there a technical term for it? And whether there is anything i should note when I'm choosing manufacturer

Thanks!

mike1808, 2023-09-25:
You have to choose a transparent material. From what I can see PCBway has an option to choose a transparent resin. However, don't expect it to be perfectly transparent since getting 100% is a challenging in 3d printing.

sedghi, 2023-09-25:
Thanks @mike1808 
I have heard it should be biocompatible resin since you touch it multiple hours per day, Do you have any comment on that too?

mike1808, 2023-09-25:
@sedghi cured resin is mostly inert. 
Here is  the video with how to order a printed part from PCBway in clear resin https://www.youtube.com/watch?v=SsvZZx0RXUY

sedghi, 2023-09-25:
Thanks a lot!

closed 2023-10-27 by braindefender