braindefender/KLP-Lame-Keycaps #14: Will we ever see the KLP-Lame as a commercial product?

alvaro-prieto, 2024-10-24:
The KLP-Lame have captured the attention of many keyboard enthusiasts, who wonder if this spectacular keyset, known for its unique and detailed design, will be released as a commercial product. Are there plans to launch these wonderful keys in PBT someday?

braindefender, 2024-11-06:
Hi. Maybe someday someone will make them in PBT. 

Personally, I don't want to dive into production/business/payment/delivery stuff. 
Not only because it is time-consuming, but also because of the sanctions imposed on Russian payment systems.

alvaro-prieto, 2024-11-07:
Thank you very much for replying! What would be necessary for a manufacturer to commercialize your profile? I know a man who has worked for a keycap manufacturing company in China, maybe you could reach some kind of agreement.