braindefender/KLP-Lame-Keycaps #17: Print settings

sadtab, 2024-11-08:
I'm trying to print these jems in PETG, I cant really find the best angle to print them, could you also include a cura or pusa project file with the optimal angle to print them, I'm wondering how you can possible print the MX version with no support

braindefender, 2024-11-09:
I don't have any particular project files for FDM, but I think supports are mandatory.
And optimal angle for Choc and MX differs, I suppose. Will try to print some MX's later

sadtab, 2024-11-11:
I appreciate if you could post some screenshots, I just can't get that perfect angle

braindefender, 2024-11-15:
But I don't think that is possible to print them entirely without supports...

closed 2024-11-15 by braindefender

braindefender, 2024-12-09:
@sadtab 

UPD: that is possible. Need post-processing, but MX socket is almost perfect and without any supports.

[image]

[KLP Lame FDM Test.zip](https://github.com/user-attachments/files/18056353/KLP.Lame.FDM.Test.zip)