braindefender/KLP-Lame-Keycaps #22: Potential issues with connectors in combined .stl files and SLS printing

alexyung, 2025-01-27:
I want to preface this by saying that I have no experience with 3D printing and modeling for 3D printing, so I'm not sure if this necessarily an issue with the model or just my choices in how I had the keycaps printed.

I recently had the combined files for the Normal MX version of these keycaps printed with SLS from JLCPCB. Unfortunately, the keycaps arrived mostly broken off their connectors already and with some pieces missing - I had 3 extra tilted caps and missing 3 thumb caps for one of the Normal Part 2 MX sets. I talked to JLC support, and got this response:

> As per your design, the wall thickness is only around 1.5mm, which will be prone to breakage and result in small parts separated so easily leads to a mix-up issue.
> To serve you better, for future orders with such kind of model, would you please add some unique letters or numbers on the keycaps or may be at the backside? Otherwise you may increase the wall thickness of the connection area to be more than 3.0mm.
> Concerning the current parts, we afraid there were some keycaps got missing. When we arranged reproduction of them, they somehow got mixed up so wrong keycaps being reprinted.

It seems they complained that the connectors were too small/weak and the caps broke off during processing, which caused the mixup.

Is it possible to make the connectors for the combined .stl's thicker/stronger to stop mixups like these from happening again? Otherwise, maybe it should be made clear what type of printing would be ideal for these keycaps?

braindefender, 2025-01-27:
Hi. Thanks for reporting this issue. I've never printed them in SLS so maybe the issue is there. I will look onto this.

braindefender, 2025-07-13:
I increased surface area of connector parts. Now they should print more stable.