braindefender/KLP-Lame-Keycaps #24: recommendations for jlc3dp printing material

SergioMG97, 2025-04-26:
Hey there, I really love the designs and I was thinking of printing a bunch of them in my next order to jlc3dp. Do you have any recomendation regarding the printing tehcnology and material, I like the of finish of the examples you published. I appreaciate any help because I'm so lost haha

braindefender, 2025-05-01:
Hi, I've never printed via JLC or any other third-party printing services.
I have both FDM and SLA printers at home, so I only used them.

So, there will be no direct recommendations. But I heard that white SLA on JLC is sun-sensitive and may gain yellow-orange tint over time.

peterjc, 2025-07-04:
I ordered the choc saddle set with JLCPLC this week, SLA resin, LC Black resin, defaults otherwise - after review both files gave:

> File Error
>
> Our system detected the following errors in your files, any other problems will be notified by email.
>
> 1. Bad edges detected
> 2. Near bad edges detected
> 3. Planar holes detected
> 4. Multi-shells detected
> 5. Possible noise shells detected
> 6. < 0.8mm wall thickness detected
> 
> There might be risks of crack, damage, deformation, loss of details.Are you able to take the risks?

and:

> Review Note:
> 1.Connected-design parts will requires more extra materials; more extra time in the post processing and package etc. So extra cost 0.1$/pcs will be charged. Also, the build time will add 1-2 more business days. For more details, please also visit our statements on the website below. Thanks ! https://jlc3dp.com/help/article/213-Connected-Parts-Printing-Guide

The price looks to have gone up by about $1 per set. Fingers crossed.

thrly, 2025-08-06:
> I ordered the choc saddle set with JLCPLC this week, SLA resin, LC Black resin, defaults otherwise - after review both files gave:
>
>
> and:
>
>
> The price looks to have gone up by about $1 per set. [...]

What were the results? I've just made an order and got the same errors...

peterjc, 2025-08-06:
Thanks for the reminder - my JLCPLC resin prints turned out nicely. 

They did seem to run into a QC issue with the keycaps through (I could login to the website and check progress - and one bundle of keycaps was clearly packaged after everything else):

> Thank you so much for ordering from JLCPCB, but we regret to tell you that your order XXX will be delayed due to some of the 3D-printing parts scrapped. It may take 2 more days to be redone/ re-manufactured for sufficient quantity for your order.

They could be removed from the choc switches much more easily than my MKB key caps (which I bought in the meantime).

Photos: 

[image]

[image]

I’m currently using the MBK set, but will build a second keyboard… https://fediscience.org/@pjacock/114870433960368787

thrly, 2025-08-06:
@peterjc thanks! Ahh interesting that they're easier to remove than MBKs (what I'm looking to replace), does that imply the fit is less secure? Wonder if that's a material or a design thing...

fantaandcode, 2025-08-09:
I'd suggest using a little bit of Kapton tape. Adhere it to the keycap and it's easy to remove later too if need be.

anrichp, 2026-02-23:
@daniel2887, I'm using JLPCB to print the keycaps but received the following feedback from the submitted files:

"The wall thickness of those indicated red areas is too thin, which has high risks of loss, deformation, crack, or damage during the printing process.

Could you please confirm if the risks are acceptable for you?

If not, could you please kindly increase it to be at least 1.0 mm to proceed? 2.0 mm will be better. Then we will activate the ' replace file' button for you."

[Image from JLPCB](https://ibb.co/G3ptrG8v)

I'm using the same MJF PA12-HP Nylon as you have previously, has JLPCB asked you these questions previously?

Thanks!