braindefender/KLP-Lame-Keycaps #25: Semi-saddle profile

arialdomartini, 2025-05-08:
What I love the saddle profile is how it is convenient for fingers to slide from the home-row forward or backward to the upper or the lower line, like on a gentle rail.

The only exception I found is for my index fingers when they have to move from their natural column (VFR MJU on a QWERTY) to the inner column (BGT NHY). 

Think to the 2 saddle borders on the right of H and the left of J

```
\___/ \___/ \___/ \___/ \___/
  H     J     K     L     ;
```

It would be nice if the borders was flat, so that the finger could more from J to H sliding.

```
\___. .___/ \___/ \___/ \___/
  H     J     K     L     ;
```

Given the symmetry, H and J can use the very same semi-saddle keycap, rotated by 180 degree.

I'm not good at CAD, so I am not sure I am able to send you a PR with such profile.

What do you think?

braindefender, 2025-05-08:
I think that is the reason why [CLP](https://github.com/vvhg1/clp-keycaps) keycaps have `Saddle Lower Side` and `Tilted Slanted` variants. It'll be good to have those on `TGB/YHN` columns. In addition, if the fingers are short enough, then moving the index finger to adjacent column results in the middle finger ending up on the edge of the keys (right on bumps). Maybe I'll consider it in the next updates.

arialdomartini, 2025-05-08:
(Just found out that https://github.com/vvhg1/clp-keycaps implemented a similar idea with the tilted+slanted keycaps. Unfortunately, they are only MX)

taropoo, 2025-07-17:
Along the lines of @arialdomartini 's request, I am wondering if @braindefender you'd consider making a less tilted version, like maybe 7 degrees? My use case is I have a lily58 layout, I am currently using saddle tilted for both first and second rows, it feels really nice to type second rows but not so much for the first row. Maybe I should just use regular saddle version for second rows 🤔