braindefender/KLP-Lame-Keycaps #26: Clarify the keycap sizes used under the Production section

floating-cat, 2025-05-12:
Hi,

I am trying to print a keycap for lapka keyboard.
But it's unclear what size is used for keycaps under the Production folder.

Could you please add details about what size is used for `[Production](https://github.com/braindefender/KLP-Lame-Keycaps/tree/master/Production)/MX/KLP_Lame_MX-Normal-4xNormal_Tilted-3xThumb-1xNormal_Homing-1xNormal-Combined.7z and KLP_Lame_MX-Normal-6xNormal_Tilted-3xNormal-Combined.7z？
Is it an MX stem with a Choc size or with an Mx size?

braindefender, 2025-05-12:
Hi, inside the `Production` folder are standard versions: MX+MX / Choc+Choc.
Currently, there are no `Production` files for MX+Choc / Choc+MX variants.

I'll add them this week (probably) with a fix from #22

braindefender, 2025-07-04:
Hi, fixed by latest commit

closed 2025-07-04 by braindefender