braindefender/KLP-Lame-Keycaps #27: Would it be possible to release bigger caps?

andywarburton, 2025-07-04:
I really love these caps and I’m seriously considering them for my daily driver. Only problem is I use 2u keys in a couple of places and it would look weird with a single key or off-brand 2u with these beautiful caps.

I don’t know about everyone else but I would really love to see a 2u version (and maybe 1.5u) in the future.

Thank you again for designing and sharing these beautiful caps!

andywarburton, 2025-08-10:
I hacked together my own 2U key. You can find it in the remixes on Printables should anyone need it.

[image]

andywarburton, 2025-08-10:
@braindefender you are welcome to use this picture in your examples if you wish!

tarikkirgin, 2025-08-13:
This looks great, what material is it?

andywarburton, 2025-08-13:
It’s Anycubic ABS like resin in clear with some green dye added